ginkage/MHI-AC-Ctrl-ESPHome #121: AC unit powers off, unable top control & getting mhi_ac_ctrl_core.loop error: -1

MadManMike12, 2025-02-10:
I have a few issues using ESPHome with ESP32 Devkit1.

- I'm  getting this in the logs: "mhi_ac_ctrl_core.loop error: -1" 
- My AC unit powers off after 60-120 seconds of being on.
- I also can't control it using ESPHome of the webpage.

I changed the framesize to 20 in the ESPHome .yaml file, this updated the sensor readings from the AC unit so I know that helped. 
I've created a fork of the project to update my SPI pinouts - this helped me get started.

Any help would be appreciated!

APIUM, 2025-02-11:
I also have the same issues using this config with a very similar board: https://gitlab.com/seanlano/mhi-ac-esp8266. I hand-soldered my first board which had it instantly, and my second board (hand placed but used paste and a heater) has developed it after a day or so being plugged in. Need to pull it out and see if I get any odd reads from the passives, will update here.

MadManMike12, 2025-02-11:
I found a couple of posts about my issues but I’m yet to resolve them. I think I might have to do some SPI diagnostics and see if I’ve done something wrong in my wiring.

AC switching off:
https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-ac-switches-power-off-sometimes

Receiving AC data, but unable to control:
https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Troubleshooting.md#fire-receiving-the-ac-status-works-but-cant-change-values

Will report back when I find something.

closed 2025-07-16 by RobertJansen1

APIUM, 2025-07-17:
My fix for this was getting a different ESP8266, whatever I was using was having the odd issues above. I had tried 2 variants but the third was the ticket.

RobertJansen1, 2025-07-17:
Which one worked and which did not?

APIUM, 2025-07-17:
They’re all on my desk at home, I’ll find out tonight and report back.

APIUM, 2025-07-18:
More difficult to find than I thought

Didn't work 
---
https://www.digikey.ca/en/products/detail/adafruit-industries-llc/2491/5761206
DigiKey Part 1528-1438-ND

Or the one I think I got from AliExpress, but I can't find it in the orders
It says on it DOTING WiFi+ CMIIT ID: 2019DP1879 FCC: https://fcc.report/FCC-ID/2AL3B-ESP-F

Works
---

The one that works came assembled from JLCPCB and it appears to be an AI Thinker ESP-12F such as this (looks the same, same CMIIT ID) https://lonelybinary.com/products/esp8266-esp-12f-module
CMIIT ID: 2019DP3307