ginkage/MHI-AC-Ctrl-ESPHome #123: ESPHome 2025.2.0 gives error on example.yaml

mhoogenbosch, 2025-02-19:
```
  The "custom" component has been removed. Consider conversion to an external component.
  https://esphome.io/guides/contributing#a-note-about-custom-components.
  platform: custom
  lambda: return ((MhiAcCtrl*)id(airco_kantoor))->get_binary_sensors();
  binary_sensors: 
    - name: Defrost
```

sanderlv, 2025-02-19:
It seems a lot has been changed in ESPhome... this will take a lot of rework (I supppose)...
https://esphome.io/guides/contributing#a-note-about-custom-components

mhoogenbosch, 2025-02-19:
> It seems a lot has been changed in ESPhome... this will take a lot of rework (I supppose)... https://esphome.io/guides/contributing#a-note-about-custom-components

Was reading up on that article too. Yes, seems so. Unfortunate.

RobertJansen1, 2025-02-19:
well, this part was already on my whishlist, but no time for it yet (#119)

basnahuis, 2025-02-19:
Now also this problem. I wanted to create a board but that doesn't work. is there also a temporary solution?
hopefully it will be corrected soon.

RobertJansen1, 2025-02-19:
Temporary solution would be to downgrade esphome, there is no fast fix for this and the heads-up from the esphome project is kinda short (I have not seen a deprecated notification for this in building for example).

mhoogenbosch, 2025-02-19:
Or just hit the ignore button for now. Nothing wrong with 2024 esphome for this device.

RobertJansen1, 2025-02-19:
> Or just hit the ignore button for now. Nothing wrong with 2024 esphome for this device.

That probably doesn't work if you want to create a new one

sanderlv, 2025-02-19:
Hmz... I'll have to have 2 instances and NOT update my test machine to be able to keep compiling...

RobertJansen1, 2025-02-19:
You can install an older version of esphome through an add-on:

https://github.com/khenderick/esphome-legacy-addons

Easier than maintaining a second home assistant instance

hberntsen, 2025-02-19:
You could also compile/upload using Docker, i.e. `docker run --rm -v "${PWD}":/config -v "${PWD}/.penv:/root/.platformio/penv" -it esphome/esphome:2024.12 run example.yaml`. Documentation here: https://esphome.io/guides/getting_started_command_line.html

RobertJansen1, 2025-02-19:
> > You can install an older version of esphome through an add-on:
> > https://github.com/khenderick/esphome-legacy-addons
> > Easier than maintaining a second home assistant instance
> 
> that doesn't work for me? do you have a better explanation how to do?

if the view in my home assistant button doesn't work (try to get it working as this is very convenient to have, most probable issue there is the wrong URL which you need to change after the first click):

Type C somewhere in your home assistant and type add-on in the search bar that appears (or go to settings > add-on store)
click the 3 dots in the right top and click on repositories
paste the URL of the repo (https://github.com/khenderick/esphome-legacy-addons) and click add

in the add-on store, the esphome legacy versions should be visible to select and install (they will be available as an additional add-on next to the standard esphome)

djansen1987, 2025-02-23:
[https://www.youtube.com/live/KyuRf8blctY?si=LQ0W1lnqCvso4QXj&t=3931](https://www.youtube.com/live/KyuRf8blctY?si=LQ0W1lnqCvso4QXj&t=3931)
it was mentioned couple of times
https://esphome.io/changelog/2025.2.0.html#removal-of-custom-components

serrj-sv, 2025-02-26:
temporary solution is to use additional external module, which brings back custom component functionality:

https://github.com/robertklep/esphome-custom-component

tested with ESPHome 2025.2.1, works fine

mhoogenbosch, 2025-02-26:
> temporary solution is to use additional external module, which brings back custom component functionality:
> 
> https://github.com/robertklep/esphome-custom-component
> 
> tested with ESPHome 2025.2.1, works fine

Nice, can you share a working config? Mine is running out into errors. Will look at it tonight to see what I need to change.

serrj-sv, 2025-02-26:
> can you share a working config

Here you go:
```
esphome:
  name: ac-lroom
  friendly_name: ${devicename}
  min_version: 2024.6.0
  platformio_options:
    # Run CPU at 160Mhz to fix mhi_ac_ctrl_core.loop error: -2
    board_build.f_cpu: 160000000L

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: INFO
  baud_rate: 0
  # logs:
  #   component: ERROR

ota:
  password: "_change_me_"
  platform: esphome

wifi:
  ssid: !secret wifi1_ssid
  password: !secret wifi1_password
    
# Version 3.0

substitutions:
  # Unique device ID in HA
  deviceid: "ac_lroom"
  # Unique MHI device ID used for climate class
  mhi_device_id: "mhi_${deviceid}"
  # Unique device name in HA (sensor names will be prefixed by this name)
  devicename: "Living Room AC"
  # If you encounter mhi_ac_ctrl_core.loop error: -2 errors, change the frame_size to 20
  frame_size: "33"

# Optionally enables the ESP web server
# web_server:
#   port: 80

external_components:
  - source: github://ginkage/MHI-AC-Ctrl-ESPHome@master
    components: [ MhiAcCtrl ]
  - source: github://robertklep/esphome-custom-component
    components: [ custom, custom_component ]

MhiAcCtrl:
  id: ${mhi_device_id}
  frame_size: ${frame_size}
  room_temp_timeout: 1800

button:
  - platform: restart
    name: Restart
    entity_category: diagnostic

api:
  reboot_timeout: 0s
  services:
    # Call the set_api_room_temperature service from HA to override the room temperature
    # If a new value has not been received after room_temp_api_timeout seconds, it will fall back to internal sensor
    - service: set_api_room_temperature
      variables:
        value: float
[... 241 lines cut ...]
    name: 3D Auto
    id: fan_control_3Dauto
    icon: mdi:video-3d
    optimistic: true
    turn_on_action:
      - lambda: |-
          if (id(frame_size_sensor).state == 33) {
            if (id(Dauto_received).state != 1) {
              ESP_LOGD("main", "Turn on 3DAuto");
              return ((MhiAcCtrl*)id(${deviceid}))->set_3Dauto(1);
            }
          } else {
            ESP_LOGD("main", "Frame size is not 33, cannot turn on 3DAuto");
          }
    turn_off_action:
      - lambda: |-
          if (id(frame_size_sensor).state == 33) {
            if (id(Dauto_received).state != 0) {
              ESP_LOGD("main", "Turn off 3DAuto");
              return ((MhiAcCtrl*)id(${deviceid}))->set_3Dauto(0);
            }
          } else {
            ESP_LOGD("main", "Frame size is not 33, cannot turn off 3DAuto");
          }
```

Freez, 2025-02-27:
It works perfect with this https://github.com/robertklep/esphome-custom-component

LawnMeower, 2025-02-27:
Hey folks, I've experimented a bit with code generation [here](https://github.com/LawnMeower/MHI-AC-Ctrl-ESPHome/tree/sensor_defs) and it seems to be working fine without external components.
It does require changes in the yaml configuration of course, but so far they are minimal.
If you'd like to try, you can use [this example](https://github.com/LawnMeower/MHI-AC-Ctrl-ESPHome/blob/sensor_defs/example.yaml) and 

```
external_components:
  - source: github://LawnMeower/MHI-AC-Ctrl-ESPHome@sensor_defs
    components: [ MhiAcCtrl ]
```

Basically sensors are part of platforms now:

```
sensor:
  - platform: MhiAcCtrl
    mhi_ac_ctrl_id: ${mhi_device_id}
    outdoor_temperature:
      name: "outdoor_temperature"
    return_air_temperature:
      name: "return_air_temperature"
```

This is not ready for the prime time yet, sensors definitions are basic and there's some unclear stuff about "old vanes positions". On the plus side now you don't have to include all the sensors in the yaml and I think we can simplify the configuration further with that vanes/3dauto stuff.

stosoorok, 2025-03-22:
Did you use platform: MhiAcCtrl? ESP installer gives an error on that line.

Second thing I noticed is very big idle power consuption, it's about 80W. Did not notice this before with 2024 ESPHome version, it was like 5W and 10-15W when Fan Only was operational. Can anyone verify it?

ajpalm, 2025-03-22:
> Second thing I noticed is very big idle power consuption, it's about 80W. Did not notice this before with 2024 ESPHome version, it was like 5W and 10-15W when Fan Only was operational. Can anyone verify it?

I've got two units, my stats are

Bigger sized unit
- standby of ~6w.
- fan only seeing 25-30w total

Smaller sized unit
- standby of ~8w.
- fan only seeing 20-25w total

So 80w seems a bit high for fan only, unless its a large unit?

RobertJansen1, 2025-04-04:
With a BIG thanks to @LawnMeower this issue is now resolved in version 4!

closed 2025-04-04 by RobertJansen1

KostyaSha, 2025-05-04:
> With a BIG thanks to [@LawnMeower](https://github.com/LawnMeower) this issue is now resolved in version 4!

could you update README with working example?

RobertJansen1, 2025-05-04:
We did update the readme, how can we make it more clear @KostyaSha ?