ginkage/MHI-AC-Ctrl-ESPHome #125: Remove the need to use an automation for external temperature sensors

paravoid, 2025-02-21:
Thank you so much for writing and maintaining this external component. It's proven invaluable to me!

I have an automation running every minute, using `set_api_room_temperature()` in order to use an external (Bluetooth) temperature sensor. I believe this is still the recommended way to do it.

First of all, it's a little bit annoying to have both custom lambdas on the ESPHome side, and an automation on the HA side. More importantly, it's also quite error-prone: I frequently see spikes in the history, presumably due to the A/C unit falling back to its internal sensor. I don't know why, but I suspect HA is just missing the heartbeats for whatever reason.

Separately to this component, I am also using another external component for units from a different vendor (their name also happens to start with "Mitsubishi" ;)), called "mitsubishi_itp", from the good folks of muart-group. They seem to have followed a bit of a different approach: https://muart-group.github.io/user/configuration/temp-sources

You basically expose the sensor from HA to ESPHome (which works automagically, with `sensor: platform: homeassistant`), and then the ESPHome component makes sure this value gets to the unit in its usual loop(). One could alternatively use other sensor platforms, such as Dallas 1-wire or a Bluetooth sensor, etc., removing the need for HA coordinating this altogether. There is even functionality to choose which source you'd like to use at runtime, through a `select`.

It'd be neat if a similar implementation took place here. Thanks again!

paravoid, 2025-03-03:
I'm experimenting with this:
```yaml
sensor:
  - platform: homeassistant
    id: ha_temperature
    name: "Home Assistant Temperature Sensor"
    entity_id: sensor.living_room_thermometer_temperature
    filters:
      - heartbeat: 30s
    on_value:
      then:
        - lambda: |-
            ((MhiAcCtrl*)id(${deviceid}))->set_room_temperature(x);
```

This updates values from HA immediately as they come in, but makes sure that even if they're unchanged, a new value is posted every 30s.

So far, so good! The spikes are gone, but more importantly, there are far fewer oscillations in temperature. With the /1min temperature resolution that I had originally configured (through an automation, as recommended), the A/C's PID was overshooting and then undershooting when there were large temperature swings, and thus struggling to keep up. With the code above, values from my external thermometer are posted immediately.

picanl, 2025-03-04:
Good alternative

binarybucks, 2025-03-28:
Agreed, this works much better than updating it from HA, thank you @paravoid!

[image]

Thank you for the hint!

ervee, 2025-03-28:
What would happen if you set it to heat to 20C and your temperatuur sensor battery dies when it gets to 18C ?

binarybucks, 2025-03-28:
Good catch, I do have notifications for that but it indeed become a problem. 

Interestingly when the HA sensor goes to "unavailable" the internal timer is reset but the stale last value is kept. Probably needs a check for setting NaN values to fallback to the internal sensor.

ervee, 2025-03-28:
Yes, IMHO there should always be a fallback or something to prevent the AC from getting out of control. On places where I depend on sensors I tend to create min/max helpers with multiple sensors. And ofcourse setup automations to check if all sensors are online.

arpiecodes, 2025-03-28:
I use a templated IF conditional to check for this exact case. Works fine and prevents the temp from being used if the sensor hasn't updated for over an hour.

[image]

`{{ (as_timestamp(now())-as_timestamp(states.sensor.klimaat_slaapkamer_temperature.last_updated)) < 14400 }}`

Sometimes the best solution is the easy one. ;-)

SuperKoe, 2025-04-23:
> I use a templated IF conditional to check for this exact case (stale temps) and updates the temp every minute if the AC is in use (other mode than OFF). Works fine and prevents the temp from being used if the sensor hasn't updated for over an hour.
> <img alt="Image" width="1258" src="https://private-user-images.githubusercontent.com/30956886/428174142-c36bce7b-6774-41eb-90ac-dc73935fe010.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU0NDM5NTIsIm5iZiI6MTc0NTQ0MzY1MiwicGF0aCI6Ii8zMDk1Njg4Ni80MjgxNzQxNDItYzM2YmNlN2ItNjc3NC00MWViLTkwYWMtZGM3MzkzNWZlMDEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDIzVDIxMjczMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg1MmMzMjQ3YmQxMmE1Mzg2Y2RlZjMzYzVmZDRhZDZkY2FjNmFkZWMyMGQ5NDg5MjYxNGYwYTFlNDRiYWZhNjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.pVvW-wnpK688_1rm4qJGz6-eZS1VvOridtoqZdYIU7w">
> 
> `{{ (as_timestamp(now())-as_timestamp(states.sensor.klimaat_slaapkamer_temperature.last_updated)) < 14400 }}`
> 
> Sometimes the best solution is the easiest one. ;-)

Can you give me the whole script? i just got my MHI ac and espboard.
The AC is pendeling alot, but the roomtemp is pretty stable so i want to send that data. but i fail to understand how.

arpiecodes, 2025-04-23:
Here's my full automation;

```
alias: Set room temperature airco beneden from klimaat beneden
description: ""
mode: single
triggers:
  - minutes: "*"
    seconds: /30
    trigger: time_pattern
conditions:
  - condition: template
    value_template: >-
      {{
      (as_timestamp(now())-as_timestamp(states.sensor.klimaat_beneden_temperature.last_updated))
      < 14400 }}
  - type: is_temperature
    condition: device
    device_id: 3f55d7bb3336f00f4fadab0c9b454803
    entity_id: sensor.klimaat_beneden_temperature
    domain: sensor
    above: 15
  - condition: not
    conditions:
      - condition: device
        device_id: 416f2875d4e33326b31b480162fcd49f
        domain: climate
        entity_id: climate.airco_beneden
        type: is_hvac_mode
        hvac_mode: "off"
actions:
  - data:
      value: "{{ states('sensor.klimaat_beneden_temperature') | float }}"
    action: esphome.mhi_beneden_set_api_room_temperature
```
I also lowered the setpoint on the unit itself. See https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/65#issuecomment-1774071728.
Note that I also set the airco in

RobertJansen1, 2025-07-16:
this was fixed by the awesome work of @LawnMeower in #133 

see [`external_sensor.yaml`](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/blob/master/examples/external_sensor.yaml) for an example on how to do this (just like @paravoid requested :))