ginkage/MHI-AC-Ctrl-ESPHome #129: Request - expose pins to be able to configure in YAML file.

Astute4185, 2025-03-15:
Hi there,

Been using a fork of a fork to make this work with an M5 Atom module and have been able to make it work.
However for ongoing usage with differing esp modules, it might be a good feature to be able to expose SCK_PIN, MOSI_PIN, MISO_PIN as a configurable item for the YAML.

My C++ and Python skill are limited, I have not really coded since Uni. So without making a massive mess this might be a feature tho add to the pot?

acfnews, 2025-04-14:
Dear Robert,

I have the same request. 

My coding skills are very limited, so most probably it is quickest if you can incorporate the 3 variables for the ESP pins?
My scenario is a ESP32 (isof ESP8266).
My esphome server is docker based, outside of my HA instance.

For now I have found the following solution/workaround based on my previous practices (before version 3.0) to work with your version 4:

1. copy a clone of github://ginkage/MHI-AC-Ctrl-ESPHome@master into my ./config directory
2. adapt the 3 variables in 'MhiAcCtrl/MHI-AC-Ctrl-core.h' on line 35 <> 37 to 
` 34 // pin defintions
 35 #define SCK_PIN  18
 36 #define MOSI_PIN 23
 37 #define MISO_PIN 19`
3. change the 'external_components' in my YAML config into:
`external_components:
  - source:
      type: local
      path: .
    components: [MhiAcCtrl]` 

So maybe there could be a default value in your code, but with a possibility to overwrite them if defined somewhere in the YAML config file?

Thanks for the great support!

RoboMagus, 2025-07-16:
@RobertJansen1, No worries. Just happy to see this implemented ;)