ginkage/MHI-AC-Ctrl-ESPHome #130: Setting an external room temperature overrides return air FB.

Hugoagogo, 2025-03-16:
Is it at all possible to still get the feedback from the internal sensor when using external temperature feedback.

I would like to trend the values for comparison, but it appears that when you start feeding external room temperatures this is overwritten.

I did attempt to have a look through the code but it wasn't immediately obvious if this was something done within this project or is just a limitation of what the air conditioner feeds back.

I am happy to work on it if somebody could give some pointers.

LawnMeower, 2025-03-28:
The external temperature that you set using API or external sensor does not directly affect the sensor value: it's the airco unit that signals return air temperature back as it was overridden. So unless there's a second undiscovered "real" return temperature value in protocol I don't think this is possible.

RobertJansen1, 2025-03-30:
We could consider storing the offset right before sending it to the unit although I don't know how a change in return air temp would behave

Hugoagogo, 2025-03-30:
It was more of there was exposed anywhere else in the frame from the aircon what the built in sensor value is.

As best I can see there is room temp and return air temperatures but both of these seem to get overridden when using the set room temperature API.

RobertJansen1, 2025-03-30:
> The external temperature that you set using API or external sensor does not directly affect the sensor value: it's the airco unit that signals return air temperature back as it was overridden. So [...]

https://github.com/absalom-muc/MHI-AC-Ctrl/issues/61#issuecomment-968312270 confirms this. getting the actual sensor value would then mean we need to temporarily disable feeding the room temp, causing the unit to go off and wander on his own sensor which is most likely not wanted.