ginkage/MHI-AC-Ctrl-ESPHome #131: No response of the unit

Rudyoost, 2025-03-17:
Hi All,

Thank you for the configuration. I've connected the board to my airco unit and it started. In HA (on my dashboard) I see that the unit has connection with HA... but, e.g. if I set the unit to cooling in HA, the airco doest start, according to HA, it thinks it turned on the unit, but nothing happens...
Also the sensors report "unknown"... 
Turning on the airco with the IR remote, the board doesn't pick that up either...
I have a SRK25ZM-S. On the PCB it states RLA505A001

Does anyone has a idea what's happening here? Or am I missing something?

thanks in advance

Ruud

ajpalm, 2025-03-17:
> I have a SRK25ZM-S. On the PCB it states RLA505A001

If you look at the logs, you'll see a heap of errors.

You need to change the frame size to 20. It covered off in the FAQ

https://github.com/ginkage/MHI-AC-Ctrl-ESPHome?tab=readme-ov-file#i-am-getting-mhi_ac_ctrl_coreloop-error--2-errors-and-nothing-works

Rudyoost, 2025-03-18:
@ajpalm 

thanks for your info. Did overlook that one. it's working now. 
But i'm having now this:

[image]

RobertJansen1, 2025-03-18:
https://github.com/ginkage/MHI-AC-Ctrl-ESPHome?tab=readme-ov-file#i-am-getting-the-following-logline-in-the-console-of-my-device