ginkage/MHI-AC-Ctrl-ESPHome #132: Airco not reachable in HA

corefarmer1, 2025-03-17:
every few hours HA loses ciontact with my MHI airco, it does have network connection.
I can not update 
I get this error When I try to clean build files or update 

INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/aircohk.yaml...
Failed config

binary_sensor.custom: [source /config/esphome/aircohk.yaml:105]
  
  The "custom" component has been removed. Consider conversion to an external component.
  https://esphome.io/guides/contributing#a-note-about-custom-components.
  platform: custom
  lambda: return ((MhiAcCtrl*)id(aircohk))->get_binary_sensors();
  binary_sensors: 
    - name: Defrost

i understand there is an error in the yaml but how can I get rid of it?
Any advice?

clickoncentral, 2025-03-17:
# Version 4.0a

substitutions:
# Unique device ID in HA
deviceid: "desk_ac"
# Unique MHI device ID used for climate class
mhi_device_id: "mhi_${deviceid}"
# Unique device name in HA (sensor names will be prefixed by this name)
devicename: ${deviceid}
# If you encounter mhi_ac_ctrl_core.loop error: -2 errors, change the
frame_size to 20
frame_size: "33"

# Optionally enables the ESP web server
# web_server:
# port: 80

external_components:
- source: ***@***.***_defs
components: [ MhiAcCtrl ]

MhiAcCtrl:
id: ${mhi_device_id}
frame_size: ${frame_size}
# Update the following to change the default room temp timeout
# room_temp_timeout: 60

button:
- platform: restart
name: Restart
entity_category: diagnostic

api:
reboot_timeout: 0s
services:
# Call the set_api_room_temperature service from HA to override the room
temperature
# If a new value has not been received after room_temp_api_timeout seconds,
it will fall back to internal sensor
- service: set_api_room_temperature
variables:
value: float
then:
- lambda: |-
return ((MhiAcCtrl*)id(${deviceid}))->set_room_temperature(value);
# Call the set_vanes service from HA to set the vane position
# Needed because the ESPHome Climate class does not support this natively
# Possible values: 1-4: static positions, 5: swing, 0: unknown
- service: set_vanes
variables:
value: int
then:
- lambda: |-
return ((MhiAcCtrl*)id(${deviceid}))->set_vanes(value);

globals:
- id: room_temp_api_timeout
type: int
restore_value: no
initial_value: "120"
- id: frame_size
type: byte
restore_value: no
initial_value: ${frame_size}

climate:
- platform: MhiAcCtrl
mhi_ac_ctrl_id: ${mhi_device_id}
id: ${deviceid}
name: "${devicename}"
[... 237 lines cut ...]
}
} else {
ESP_LOGD("main", "Frame size is not 33, cannot turn on 3DAuto");
}
turn_off_action:
- lambda: |-
if (id(frame_size_sensor).state == 33) {
if (id(Dauto_received).state != 0) {
ESP_LOGD("main", "Turn off 3DAuto");
return ((MhiAcCtrl*)id(${deviceid}))->set_3Dauto(0);
}
} else {
ESP_LOGD("main", "Frame size is not 33, cannot turn off 3DAuto");
}

*Thank you.*

*Please be aware that this communication is intended only for the recipient
and may contain confidential and legally privileged information. The views
expressed in this email are solely those of the author and do not
necessarily represent those of Click On Central. Any unauthorised review,
retransmission, dissemination, reliance on, or other use of this
communication is strictly prohibited. If you have received this email in
error, please notify Click On Central immediately by returning the email
and deleting all copies from your system.*

corefarmer1, 2025-03-17:
wow tot was quick thank you 
wil instal asap

ajpalm, 2025-03-17:
> The "custom" component has been removed. Consider conversion to an external component.

Dup of this issue https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/123

corefarmer1, 2025-03-18:
i am so sorry have been working on youre code but get nothing but errors, probably becouse i am not a coder and copy paste does not work.
i under stand i have to put in my on device name 
what about the external component device name.
and than the spaces in front of the code drive me nuts

Any tips ?

clickoncentral, 2025-03-19:
> i am so sorry having a realy hard time using your code
> becouse i have trouble putting the right amount of spaces before each line.
> is there any way you can post it with the rigt amount of spaces?

Yaml file attached.

See https://github.com/LawnMeower/MHI-AC-Ctrl-ESPHome/tree/sensor_defs

*Thank you.*

*Please be aware that this communication is intended only for the recipient
and may contain confidential and legally privileged information. The views
expressed in this email are solely those of the author and do not
necessarily represent those of Click On Central. Any unauthorised review,
retransmission, dissemination, reliance on, or other use of this
communication is strictly prohibited. If you have received this email in
error, please notify Click On Central immediately by returning the email
and deleting all copies from your system.*

corefarmer1, 2025-03-19:
Thank you any reason why this code fails (for me) 

INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/aircohk.yaml...
INFO Unable to import component MhiAcCtrl.binary_sensor: No module named 'esphome.components.MhiAcCtrl.binary_sensor'
Failed config

binary_sensor.MhiAcCtrl: [source /config/esphome/aircohk.yaml:107]
  
  Platform not found: 'binary_sensor.MhiAcCtrl'.
  platform: MhiAcCtrl
  mhi_ac_ctrl_id: mhi_aircohk
  defrost: 
    name: Defrost

clickoncentral, 2025-03-19:
Did you leave this line in?

external_components:
  - source: ***@***.***
    components: [ MhiAcCtrl ]

*Thank you.*

*Please be aware that this communication is intended only for the recipient
and may contain confidential and legally privileged information. The views
expressed in this email are solely those of the author and do not
necessarily represent those of Click On Central. Any unauthorised review,
retransmission, dissemination, reliance on, or other use of this
communication is strictly prohibited. If you have received this email in
error, please notify Click On Central immediately by returning the email
and deleting all copies from your system.*

corefarmer1, 2025-03-19:
this is in the example file 
external_components:
  - source: ***@***.***
    components: [ MhiAcCtrl ]

shoud i swap it for ?

external_components:
  - source: ***@***.***
    components: [ MhiAcCtrl ]

clickoncentral, 2025-03-19:
You need these lines for the components:

external_components:
  - source: ***@***.***
    components: [ MhiAcCtrl ]

*Thank you.*

*Please be aware that this communication is intended only for the recipient
and may contain confidential and legally privileged information. The views
expressed in this email are solely those of the author and do not
necessarily represent those of Click On Central. Any unauthorised review,
retransmission, dissemination, reliance on, or other use of this
communication is strictly prohibited. If you have received this email in
error, please notify Click On Central immediately by returning the email
and deleting all copies from your system.*

RobertJansen1, 2025-04-04:
#133 should fix this, so please update (replace?) your yaml with the new examples provided and it should compile on esphome 2025.2 and newer again

corefarmer1, 2025-04-16:
any idea's on why I get this error while uploading 
collect2: error: ld returned 1 exit status
*** [.pioenvs/aircohk/firmware.elf] Error 1
`# ginkage MHI AC Ctrl ESPHome 4.0 4-2025

esphome:
  name: aircohk
  friendly_name: aircohk
  min_version: 2024.6.0
  platformio_options:
    # Run CPU at 160Mhz to fix mhi_ac_ctrl_core.loop error: -2
    board_build.f_cpu: 160000000L

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: INFO
  baud_rate: 0
  # logs:
  #   component: ERROR

ota:
  - platform: esphome
    password: !secret ota_password 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${devicename}
    password: xfxxfxfxfxuhjx

# Optionally enables the ESP web server
web_server:
  port: 80

captive_portal:

external_components:
  - source: github://ginkage/MHI-AC-Ctrl-ESPHome@master
    components: [MhiAcCtrl]

api:
  encryption:
    key: "6uJ//8H6JwhoQXOi7NT8BdErd6ojfrovjev9ejkgKeEupsk="

  reboot_timeout: 0s
  services:
    - service: set_vertical_vanes
      variables:
        position_value: int # 1-4: static positions, 5: swing, 0: unknown
      then:
        - climate.mhi.set_vertical_vanes:
            position: !lambda "return position_value;"
    - service: set_horizontal_vanes
      variables:
        position_value: int # 1-7: static positions, 8: swing, 0: unknown
      then:
        - climate.mhi.set_horizontal_vanes:
            position: !lambda "return position_value;"
    - service: set_external_room_temperature
      variables:
        temperature_value: float # temperature to set in Celsius
      then:
        - climate.mhi.set_external_room_temperature:
            temperature: !lambda "return temperature_value;"

# Version 4.0

MhiAcCtrl:
  # Only 20 (legacy) or 33 (includes 3D auto and vertical vanes) possible.
  # If you encounter mhi_ac_ctrl_core.loop error: -2 errors, change the frame_size to 20
  frame_size: 33
  initial_vertical_vanes_position: 5
  initial_horizontal_vanes_position: 8
  # Update the following to change the default room temp timeout
  room_temp_timeout: 60

button:
  - platform: restart
    name: Restart
    entity_category: diagnostic

climate:
  - platform: MhiAcCtrl
    name: "MHI Air Conditioner"
    visual:
      temperature_step:
        target_temperature: 0.5
        current_temperature: 0.1     

time:
  - platform: homeassistant
    id: homeassistant_time

binary_sensor:
  - platform: MhiAcCtrl
    defrost:
      name: "Defrost"
    vanes_3d_auto_enabled:
      name: "3D Auto"

sensor:

#external temperature sensor
  - platform: homeassistant
    id: room_temperature_sensor
    entity_id: sensor.temperatuur_display_huiskamer_temperature 
  - platform: uptime
    name: Uptime
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s
  - platform: MhiAcCtrl
    outdoor_temperature:
      name: "Outdoor temperature"
    return_air_temperature:
      name: "Return air temperature"
    outdoor_unit_fan_speed:
      name: "Outdoor unit fan speed"
    indoor_unit_fan_speed:
      name: "Indoor unit fan speed"
    compressor_frequency:
      name: "Compressor frequency"
    indoor_unit_total_run_time:
      name: "Indoor unit total run time"
    compressor_total_run_time:
      name: "Compressor total run time"
    current_power:
      name: "Current power"
    energy_used:
      name: "Energy used"
    indoor_unit_thi_r1:
      name: "Indoor (U-bend) HE temp 1"
    indoor_unit_thi_r2:
      name: "Indoor (capillary) HE temp 2"
    indoor_unit_thi_r3:
      name: "Indoor (suction header) HE temp 3"
    outdoor_unit_tho_r1:
      name: "Outdoor HE temp"
    outdoor_unit_expansion_valve:
      name: "Outdoor unit exp. valve"
    outdoor_unit_discharge_pipe:
      name: "Outdoor unit discharge pipe"
    outdoor_unit_discharge_pipe_super_heat:
      name: "Outdoor unit discharge pipe super heat"
    protection_state_number:
      name: "Compressor protection code"
    error_code:
      name: "Error code"
    vanes_pos:
      name: "Vanes"
    vanesLR_pos:
      name: "Vanes Left/Right"

text_sensor:
  - platform: version
    name: ESPHome Version
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID
  - platform: MhiAcCtrl
    protection_state:
      name: "Compressor protection status"

select:
  - platform: MhiAcCtrl
    vertical_vanes:
      name: Fan Control Up Down
    horizontal_vanes:
      name: Fan Control Left Right

switch:
  - platform: MhiAcCtrl
    vanes_3d_auto:
      name: "3D Auto"

RobertJansen1, 2025-04-16:
can you paste the full log and yaml in code tags so it is readable?

corefarmer1, 2025-04-18:
I partly fixed the problem 
i made a new esphome device in Home assistant and managed to upload the code, Witch looks really nice.
But now my airco does not respond at all, eventhow home assistant sees the new device

RobertJansen1, 2025-07-08:
https://github.com/ginkage/MHI-AC-Ctrl-ESPHome?tab=readme-ov-file#i-am-getting-mhi_ac_ctrl_coreloop-error--2-errors-and-nothing-works

this is probably because you have an older unit, you should configure the smaller framesize

closed 2025-07-08 by RobertJansen1