ginkage/MHI-AC-Ctrl-ESPHome #138: Question about api/services/service: set_external_room_temperature

locussss, 2025-04-04:
Hi,

In the full.yaml:

    - service: set_external_room_temperature
      variables:
        temperature_value: int # 1-7: static positions, 8: swing, 0: unknown
      then:
        - climate.mhi.set_external_room_temperature:
            temperature: !lambda "return temperature_value;"

The temperature_value is _int_ with the comment of the set_horizontal_vanes. Is that right?

Shouldn't the temperature_value be a float?

RobertJansen1, 2025-04-07:
thanks for reporting, this is now fixed: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/commit/6d440399038f809ef1eacb6b3c24906da6cd9332

closed 2025-04-07 by RobertJansen1

LawnMeower, 2025-04-08:
Fixed here: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/142/commits

LawnMeower, 2025-04-09:
@locussss  

Try to put  `refresh: 0s` under external component import like this:
``` yaml
external_components:
  - source: github://ginkage/MHI-AC-Ctrl-ESPHome@master
    components: [MhiAcCtrl]
    refresh: 0s
```

This is to tell ESPHome and git to fetch latest version

Also you can test the action using action tab in HA developer tools.

locussss, 2025-04-09:
Yes!

The refresh plus clean build files and install did the trick. Thank you.

[image]