ginkage/MHI-AC-Ctrl-ESPHome #139: sensor.h error

Bill-Dung, 2025-04-05:
I get following error when I try to compile Version 4.0. Clean build was without success. All other ESPHome devices are compiling without problems.

`INFO ESPHome 2025.3.3
INFO Reading configuration /config/esphome/test-ac.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test-ac (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling .pioenvs/test-ac/src/esphome/components/MhiAcCtrl/climate/mhi_climate.cpp.o
Compiling .pioenvs/test-ac/src/esphome/components/MhiAcCtrl/mhi_platform.cpp.o
Compiling .pioenvs/test-ac/src/esphome/components/MhiAcCtrl/select/mhi_horizontal_vanes_select.cpp.o
Compiling .pioenvs/test-ac/src/esphome/components/MhiAcCtrl/select/mhi_vertical_vanes_select.cpp.o
In file included from src/esphome/components/MhiAcCtrl/mhi_platform.cpp:1:
src/esphome/components/MhiAcCtrl/mhi_platform.h:5:10: fatal error: esphome/components/sensor/sensor.h: No such file or directory
    5 | #include "esphome/components/sensor/sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test-ac/src/esphome/components/MhiAcCtrl/mhi_platform.cpp.o] Error 1
In file included from src/esphome/components/MhiAcCtrl/climate/mhi_climate.h:6,
                 from src/esphome/components/MhiAcCtrl/climate/mhi_climate.cpp:2:
src/esphome/components/MhiAcCtrl/climate/../mhi_platform.h:5:10: fatal error: esphome/components/sensor/sensor.h: No such file or directory
    5 | #include "esphome/components/sensor/sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test-ac/src/esphome/components/MhiAcCtrl/climate/mhi_climate.cpp.o] Error 1
In file included from src/esphome/components/MhiAcCtrl/select/mhi_horizontal_vanes_select.h:3,
                 from src/esphome/components/MhiAcCtrl/select/mhi_horizontal_vanes_select.cpp:2:
src/esphome/components/MhiAcCtrl/select/../mhi_platform.h:5:10: fatal error: esphome/components/sensor/sensor.h: No such file or directory
    5 | #include "esphome/components/sensor/sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test-ac/src/esphome/components/MhiAcCtrl/select/mhi_horizontal_vanes_select.cpp.o] Error 1
In file included from src/esphome/components/MhiAcCtrl/select/mhi_vertical_vanes_select.h:4,
                 from src/esphome/components/MhiAcCtrl/select/mhi_vertical_vanes_select.cpp:2:
src/esphome/components/MhiAcCtrl/select/../mhi_platform.h:5:10: fatal error: esphome/components/sensor/sensor.h: No such file or directory
    5 | #include "esphome/components/sensor/sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test-ac/src/esphome/components/MhiAcCtrl/select/mhi_vertical_vanes_select.cpp.o] Error 1
========================== [FAILED] Took 0.74 seconds ==========================`

Any idea?

Bill-Dung, 2025-04-05:
Here you are. This is the last version I tried. 

`esphome:
  name: test-ac
  friendly_name: AC Test
  min_version: 2024.6.0
  platformio_options:
    # Run CPU at 160Mhz to fix mhi_ac_ctrl_core.loop error: -2
    board_build.f_cpu: 160000000L

esp8266:
  board: d1_mini

logger:
  level: INFO
  baud_rate: 0

ota:
  - platform: esphome
    password: ********

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

api:

external_components:
  - source: github://ginkage/MHI-AC-Ctrl-ESPHome@master
    components: [MhiAcCtrl]

# Version 4.0
MhiAcCtrl:
  # Only 20 (legacy) or 33 (includes 3D auto and vertical vanes) possible.
  # If you encounter mhi_ac_ctrl_core.loop error: -2 errors, change the frame_size to 20
  frame_size: 33

climate:
  - platform: MhiAcCtrl
    name: "MHI Air Conditioner"

select:
  - platform: MhiAcCtrl
    vertical_vanes:
      name: Fan Control Up Down
    horizontal_vanes:
      name: Fan Control Left Right

switch:
  - platform: MhiAcCtrl
    vanes_3d_auto:
      name: "3D Auto"`

Bill-Dung, 2025-04-05:
Now I discovered that the problem only exists with the simple.yaml, the full.yaml works fine.

XMaarten, 2025-04-05:
I see the same issue, adding a single sensor entry to simply.yaml fixes the error message too.

RobertJansen1, 2025-04-07:
thanks for reporting the issue @Bill-Dung and finding out the issue @XMaarten this is now fixed in master https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/commit/31e93a229fc6d67cfc0a8275ac4f70f00fe96874

closed 2025-04-07 by RobertJansen1