ginkage/MHI-AC-Ctrl-ESPHome #141: set external temperature not working anymore?

Theetjuh, 2025-04-07:
Setting a temperature doesn’t seem to work anymore?

Executed: April 7, 2025 at 22:52:59
**Error: extra keys not allowed @ data['value']**
Result:
params:
  domain: esphome
  service: mhi_landing_set_external_room_temperature
  service_data:
    value: 20.09
  target: {}
running_script: false

I am using the same automation as before the change to v4.0, just changed ‘api’ to ‘external’.

arpiecodes, 2025-04-07:
Possibly running an older version? Since there was a fix pushed recently for something related; https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/commit/6d440399038f809ef1eacb6b3c24906da6cd9332

Theetjuh, 2025-04-07:
> Possibly running an older version? Since there was a fix pushed recently for something related; [6d44039](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/commit/6d440399038f809ef1eacb6b3c24906da6cd9332)

Yeah saw that one, but no got ‘float’ in the yaml

arpiecodes, 2025-04-07:
It's actually the other way around; the variable `temperature_value` should become `float`, not `int`. 

You could also try setting `temperature_value` instead of `value` under `service_data`, as that seems to have been renamed (no idea why though, seems to have been a very unnecessary change to me).

Theetjuh, 2025-04-07:
> It's actually the other way around; the variable `temperature_value` should become `float`, not `int`. But I guess if you include the external component from GitHub it should update it automatically once you re-upload the project within ESPHome.
> 
> You could also try setting `temperature_value` instead of `value` under `service_data`, as that seems to have been renamed (no idea why though, seems to have been a very unnecessary change to me).

That was it, when I changed ‘value’ to ‘temperature_value’ that fixed it!

Thanks 🙏

closed 2025-04-07 by Theetjuh