ginkage/MHI-AC-Ctrl-ESPHome #143: v4.0: Can't compile

TahoeNL, 2025-04-11:
I have converted my old version yamlo files to the new, using the full.yaml example file and adding the required field from the previous, like API key etc. Now the firmware won't compile and throws there errors:

```
Linking .pioenvs/aircoslaapkamer/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_Z13calc_checksumPh':
MHI-AC-Ctrl-core.cpp:(.text._Z13calc_checksumPh+0x0): multiple definition of `_Z13calc_checksumPh'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._Z13calc_checksumPh+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_Z20calc_checksumFrame33Ph':
MHI-AC-Ctrl-core.cpp:(.text._Z20calc_checksumFrame33Ph+0x0): multiple definition of `_Z20calc_checksumFrame33Ph'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._Z20calc_checksumFrame33Ph+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core16reset_old_valuesEv':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core16reset_old_valuesEv+0x8): multiple definition of `_ZN16MHI_AC_Ctrl_Core16reset_old_valuesEv'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core16reset_old_valuesEv+0x8): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core4initEv':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core4initEv+0x10): multiple definition of `_ZN16MHI_AC_Ctrl_Core4initEv'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core4initEv+0x10): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core9set_powerEb':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core9set_powerEb+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core9set_powerEb'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core9set_powerEb+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core8set_modeE6ACMode':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core8set_modeE6ACMode+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core8set_modeE6ACMode'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core8set_modeE6ACMode+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core13set_tsetpointEj':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core13set_tsetpointEj+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core13set_tsetpointEj'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core13set_tsetpointEj+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core7set_fanEj':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core7set_fanEj+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core7set_fanEj'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core7set_fanEj+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core10set_3DautoE8AC3Dauto':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core10set_3DautoE8AC3Dauto+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core10set_3DautoE8AC3Dauto'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core10set_3DautoE8AC3Dauto+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core9set_vanesEj':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core9set_vanesEj+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core9set_vanesEj'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core9set_vanesEj+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core11set_vanesLREj':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core11set_vanesLREj+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core11set_vanesLREj'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core11set_vanesLREj+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core17request_ErrOpDataEv':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core17request_ErrOpDataEv+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core17request_ErrOpDataEv'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core17request_ErrOpDataEv+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core9set_troomEh':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core9set_troomEh+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core9set_troomEh'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core9set_troomEh+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core16get_troom_offsetEv':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core16get_troom_offsetEv+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core16get_troom_offsetEv'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core16get_troom_offsetEv+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core16set_troom_offsetEf':
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core16set_troom_offsetEf+0x0): multiple definition of `_ZN16MHI_AC_Ctrl_Core16set_troom_offsetEf'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core16set_troom_offsetEf+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/aircoslaapkamer/src/esphome/components/MhiAcCtrl/MHI-AC-Ctrl-core.cpp.o: in function `_ZN16MHI_AC_Ctrl_Core14set_frame_sizeEh':
[... cut ...]
MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core4loopEj+0x80): multiple definition of `_ZN16MHI_AC_Ctrl_Core4loopEj'; .pioenvs/aircoslaapkamer/src/MHI-AC-Ctrl-core.cpp.o:MHI-AC-Ctrl-core.cpp:(.text._ZN16MHI_AC_Ctrl_Core4loopEj+0x80): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/aircoslaapkamer/firmware.elf] Error 1
========================= [FAILED] Took 52.86 seconds =========================
```

I have tried cleaning build files; but doesn't help. Any idea what might be wronng?

RobertJansen1, 2025-04-11:
i have seen a number of times before in issues cleaning build files not always completely cleans all stuff, creating a new device in esphome fixes this as that creates a new directory

TahoeNL, 2025-04-11:
Ah, hard removing the build directory did the trick. Took a while to find/do it though. Thanx.