ginkage/MHI-AC-Ctrl-ESPHome #144: Using external room_temp_sensor spams the log

hmmbob, 2025-04-15:
Today I started to migrate to the new version.
I tried adding the homeassistant sensor based way of setting an external room temperature sensor as per the [docs](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/blob/master/examples/external_sensor.yaml), but then the log file of the device goes haywire:

```
[10:56:03][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:03][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:03][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:03][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:03][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:03][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:03][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:03][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:04][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:05][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
[10:56:06][D][mhi.platform:087]: set room_temp_api: 20.799999 144 144
```

Using this yaml:
```yaml
substitutions:
  devicename: ${deviceid} 
  device_description: ESPHome MHI-AC-Ctrl-ESPHome
  deviceid: "airco_woonkamer"
  friendly_name: "Airco woonkamer"

################ Generics ################
esphome:
  name: ${deviceid}
  friendly_name: ${friendly_name}
  comment: "${device_description}"
  platformio_options:
    # Run CPU at 160Mhz to fix mhi_ac_ctrl_core.loop error: -2
    board_build.f_cpu: 160000000L

esp8266:
  board: d1_mini

# General settings: captive portal, logger, api, ota, time, restart switch, wifi
# General sensors: status, uptime, version, wifi info etc
packages:
  general_settings: !include includes/general_settings.yaml
  general_sensors: !include includes/general_sensors.yaml

web_server:

################ Components ################
external_components:
  - source: github://ginkage/MHI-AC-Ctrl-ESPHome@master
    components: [MhiAcCtrl]

MhiAcCtrl:
  external_temperature_sensor: room_temperature_sensor
  frame_size: 33
  initial_horizontal_vanes_position: 2
  initial_vertical_vanes_position: 5
  room_temp_timeout: 1800

api:
  reboot_timeout: 0s
  services:
    - service: set_vertical_vanes
      variables:
        position_value: int # 1-4: static positions, 5: swing, 0: unknown
      then:
        - climate.mhi.set_vertical_vanes:
            position: !lambda "return position_value;"
    - service: set_horizontal_vanes
      variables:
        position_value: int # 1-7: static positions, 8: swing, 0: unknown
      then:
        - climate.mhi.set_horizontal_vanes:
            position: !lambda "return position_value;"
    - service: set_external_room_temperature
      variables:
        temperature_value: float # temperature to set in Celsius
      then:
        - climate.mhi.set_external_room_temperature:
            temperature: !lambda "return temperature_value;"

climate:
  - platform: MhiAcCtrl
    name: "${friendly_name}"
    visual:
      temperature_step:
        target_temperature: 0.5
        current_temperature: 0.1

sensor:
  - platform: homeassistant #external temperature sensor
    id: room_temperature_sensor
    entity_id: climate.woonkamer
    attribute: current_temperature

  - platform: MhiAcCtrl
    outdoor_temperature:
      name: "Outdoor temperature"
    return_air_temperature:
      name: "Return air temperature"
    outdoor_unit_fan_speed:
      name: "Outdoor unit fan speed"
    indoor_unit_fan_speed:
      name: "Indoor unit fan speed"
    compressor_frequency:
      name: "Compressor frequency"
      disabled_by_default: true
    indoor_unit_total_run_time:
      name: "Indoor unit total run time"
    compressor_total_run_time:
      name: "Compressor total run time"
    current_power:
      name: "Current power"
    energy_used:
      name: "Energy used"
    error_code:
      name: "Error code"
    ### Disabled by default, but can be enabled if needed
    indoor_unit_thi_r1:
      name: "Indoor (U-bend) HE temp 1"
      disabled_by_default: true
    indoor_unit_thi_r2:
      name: "Indoor (capillary) HE temp 2"
      disabled_by_default: true
    indoor_unit_thi_r3:
      name: "Indoor (suction header) HE temp 3"
      disabled_by_default: true
    outdoor_unit_tho_r1:
      name: "Outdoor HE temp"
      disabled_by_default: true
    outdoor_unit_expansion_valve:
      name: "Outdoor unit exp. valve"
    outdoor_unit_discharge_pipe:
      name: "Outdoor unit discharge pipe"
      disabled_by_default: true
    outdoor_unit_discharge_pipe_super_heat:
      name: "Outdoor unit discharge pipe super heat"
      disabled_by_default: true
    protection_state_number:
      name: "Compressor protection code"
      disabled_by_default: true
    vanes_pos:
      name: "Vanes"
      disabled_by_default: true
    vanesLR_pos:
      name: "Vanes Left/Right"
      disabled_by_default: true

select:
  - platform: MhiAcCtrl
    vertical_vanes:
      name: Fan Control Up Down
    horizontal_vanes:
      name: Fan Control Left Right

switch:
  - platform: MhiAcCtrl
    vanes_3d_auto:
      name: "3D Auto"

text_sensor:
  - platform: MhiAcCtrl
    protection_state:
      name: "Compressor protection status"
      disabled_by_default: true
```

Am I doing something wrong?

hmmbob, 2025-04-16:
But, I'm not sending anything... The esphome code is retrieving the value directly from Home Assistant, so I'm not sure what I can do?

Why did you close the issue?

hmmbob, 2025-04-16:
Maybe @LawnMeower has an idea on why this is happening?

To add, putting `{{ state_attr('climate.woonkamer','current_temperature') }}` in the developer tools in HA gives me a nice `20.5` as output, so I'm not sure at all where the high number of decimals is coming from

[image]

LawnMeower, 2025-04-16:
Well, this is sideeffect of one of my changes for external sensor, long story short: we're setting the external temperature on every loop, and when I introduced external sensor it seemed to be the most reliable way to just grab the sensor value and set it, since we always communicate this value to AC unit.

I kept the debug log because it's debug, useful when you're trying to figure out issues, you're not expected to use it in production.  
You can disable debug logs, for instance:
```
logger:
  level: INFO

```

@RobertJansen1 Do you think it's something worth changing?

hmmbob, 2025-04-16:
That explanation makes sense. I'm not sure if we're 'stressing the ESP device too much' if we set it so often. It 'feels' busy when I upload the file with the external sensor enabled (i.e. slower/sluggish responses), but not sure if that is an actual measurable thing.

LawnMeower, 2025-04-16:
We don't stress ESP while setting the temp often, because it's part of the package that we send to AC every loop anyway, we send it with temperature or without, same workload.
What's much more stressing is the logging itself, but again: you can disable it. Or we can remove logging of the temperature override.

More context here: Initially I was playing with subscribing to the sensor updates, but this is more tricky because we also have API actions and timeouts and I found this solution unstable. So now we have easier solution that sets the temp every loop, but a bit noisier in debug logs. Which could be useful :)

hmmbob, 2025-04-16:
I made it more specific (because I actually do like to see the other debug info), and this works fine:

```yaml
logger:
  logs:
    mhi.platform: INFO
```

[image]

closed 2025-07-16 by RobertJansen1