ginkage/MHI-AC-Ctrl-ESPHome #145: V4.0 Indoor (capillary) HE temp 2 appears incorrect (maybe degrees F instead of C?)

ghenry22, 2025-04-18:
See screenshot attached, other temp sensors seem about right.  The unit is powered off, In a powered off state I doubt that any component inside the unit is sitting at 71 degrees celsius.  If I assume that value is in fahrenheit and convert it to celsius then it looks right though.

I just can't seem a powered off system having components sitting there at that temp, it would be a bit mad.

[image]

ajpalm, 2025-04-18:
I've noticed this as well, but as I have two ME AC units and they both are reporting the same 71.66c value for HE temp 2 like you when turned off.

I assumed it is a default value the units use. 

I've uploaded my two units, and you'll see lots of values are the same on both, HE temp 2, HE temp 3, discharge pipe.

So I wouldn't be too worried about it.

[image]

[image]

ghenry22, 2025-04-18:
Actually just noticed that the value doesn't change at all when running in cool / heat / off.  So perhaps it is picking up an incorrect value or something which has no real meaning?  Or perhaps it just gets a default value as you say unless there is some specific circumstance.

Thought I'd log it anyway just for reference if nothing else.

RobertJansen1, 2025-04-18:
and 71.66 is the highest value i have seen reported in this attribute in the past (also in older versions) , so the value might very well just be valid.

ajpalm, 2025-04-18:
> Actually just noticed that the value doesn't change at all when running in cool / heat / off

Not sure how the value is set, but mine moves around on its own even when the unit hasn't been running. So this value might be a calculation based on some other values.

[image]

RobertJansen1, 2025-07-08:
i checked the source code of absalom-muc and the formula for this calculation is not known:
https://github.com/absalom-muc/MHI-AC-Ctrl/blob/1241877872caa1913f0a425279323c948a94a3ad/src/MHI-AC-Ctrl.ino#L387C1-L388C89
this is not a v4.0 specific problem, just something noone found out yet :(

lvschouwen, 2026-09-27:
Some data on this from my fork of absalom-muc/MHI-AC-Ctrl (https://github.com/lvschouwen/MHI-AC-Ctrl), which now publishes the raw byte next to the converted value (request `0x40 0x81`, DB11). Setup: two SRK20ZS-WF indoor units on one multi-split outdoor unit; both indoor units have an EEV.

**At rest** (both units off, compressor 0 Hz, outdoor air 20.5 °C, rooms 20–21.5 °C): raw **252–254**.
- The old `0.327 × raw − 11.4` gives ~71 °C. That is exactly the 71.66 °C in the original report, so that reading is simply the old formula applied to a byte near its ceiling, not a swapped sensor.
- `0.275 × raw − 47` (v4.2) gives ~22 °C, which looks right at rest. The run below shows that is probably a coincidence.

**Run, raw byte ranges:**

| Phase | THI-R2 (0x40 0x81) | THO-R1 (0x40 0x82) | THI-R1 / R3 (0xc0 0x81 / 0x87) |
|---|---|---|---|
| Rest, ~21 °C | 252–254 | 60–64 | 72–81 |
| Cooling, 35–74 Hz | **254, never moved** | 62–120 | 36–73 |
| Heating, 40–75 Hz | 218–250 | 35–47 | 106–225 |

- **THI-R2** stayed flat at 254 through 20 minutes of cooling and only moved after the 4-way valve reversed into heating. It does not behave like an indoor coil sensor. It looks like a sensor that is either absent on these units or sits on the outdoor/suction side and saturates in cooling. That fits phoenix-blue's point that indoor units with an EEV may have no capillary sensor, and it fits the request prefix `0x40`, which the other outdoor values use.
- **THO-R1** behaves like the outdoor coil: hot in cooling (up to 120 raw), cold in heating (35–47). So on these units the two are not swapped, but THO-R1's formula is unknown: `0.327 × raw − 11.4` gives 8–9 °C at rest with 20.5 °C outdoor air.
- **THI-R1 / R3** rise with temperature as expected. The old linear formula looks plausible in cooling and heating but reads ~6 °C low at rest, so the real curve is probably a nonlinear NTC.

No reference thermometer was on the coils, so none of this fits a formula yet, but it may help anyone who has one. Happy to share more captures.

blues-sechseck, 2026-09-27:
I maintain the Home Assistant integration for the WF-RAC module, which reads the same operation data through the module, so your raw bytes line up with ours. We fitted the coil channels against a thermometer: they are a ~5.2 kΩ NTC (B ≈ 3900) in a divider, `raw = 367 · Rs / (Rs + R(T))`, with **Rs = 1912 Ω** for THI-R1/R3 and roughly **1230 Ω** for THO-R1.

On your rest values that gives 20.7–22.4 °C for THO-R1 against 20.5 °C outdoor air, and 72–73 °C for THI-R2 at 252–254. So the 71.66 °C from the start of this issue is a channel sitting at its ceiling, as you suspected. Your THI-R1/R3 rest values come out at 16–19 °C, a bit low for a 20–21.5 °C room – had the units been cooling shortly before?

The inverse formula and the constants are in §5.4 of https://github.com/blues-sechseck/Mitsubishi-WF-RAC-Integration/blob/main/docs/wf-rac-module-reference.md. Please note: above raw 170 the indoor curve is extrapolated.