ginkage/MHI-AC-Ctrl-ESPHome #146: Question web interface

Rudyoost, 2025-04-21:
Hi,

First of all, great job on this piece of code.
question though, in the picture you can see, that one AC has the option to "visit" the/a web interface (which works) but the other doesn't. The one that's not visitable is compiled with the current version of full yaml from this repository... what can that be?
thanks in advance.

[image]

RobertJansen1, 2025-04-21:
The full yaml does not have the webserver settings by default, that could explain the difference

Rudyoost, 2025-04-21:
Oke clear, thanks Robert. I need to figure out what i did with the frist one, that does have the web server on it.

side question though, i have another device that can be updated via ESPHome... can something be created for this script as well?

thanks again.

Rudyoost, 2025-04-23:
Thanks again Robert.
Fixed the web server part as well.

closed 2025-04-23 by Rudyoost