ginkage/MHI-AC-Ctrl-ESPHome #147: External Temperature not correct

Alfagek, 2025-04-25:
With version 4.0 i do get a wrong value's from the external temp sensor in the climate sensor.

For example:

External Temp sensor is 19.97. 
Climate Sensor is saying: 19.8.

Or

External Temp sensor is 19.57
Climate Sensor is saying: 19.2

Or like the screenshot below:

External Temp sensor: 19.63
Climate Sensor is saying: 19.5

[image]

[image]

[image]

````````````
esphome:
  name: ${deviceid}
  friendly_name: ${devicename}
  min_version: 2024.6.0
  platformio_options:
    # Run CPU at 160Mhz to fix mhi_ac_ctrl_core.loop error: -2
    board_build.f_cpu: 160000000L

esp8266:
  board: d1_mini

# Enable logging
logger:
  logs:
    mhi.platform: INFO
#  level: INFO
# baud_rate: 0
  # logs:
  #   component: ERROR

ota:
  password: <PLEASE SUPPLY PASSWORD>
  platform: esphome

wifi:
  ssid: !secret wifi_ssid32
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${devicename}
    password: $secret fallback_password

captive_portal:
    
substitutions:
  # Unique device ID in HA
  deviceid: "airco_beneden"
  # Unique device name in HA (sensor names will be prefixed by this name)
  devicename: "Airco Beneden"
  # Unique MHI device ID used for climate class

# Optionally enables the ESP web server
# web_server:
#   port: 80

external_components:
  - source: github://ginkage/MHI-AC-Ctrl-ESPHome@master
    components: [ MhiAcCtrl ]

api:
  reboot_timeout: 0s
  services:
    - service: set_vertical_vanes
      variables:
        position_value: int # 1-4: static positions, 5: swing, 0: unknown
      then:
        - climate.mhi.set_vertical_vanes:
            position: !lambda "return position_value;"
    - service: set_horizontal_vanes
      variables:
        position_value: int # 1-7: static positions, 8: swing, 0: unknown
      then:
        - climate.mhi.set_horizontal_vanes:
            position: !lambda "return position_value;"
    - service: set_external_room_temperature
      variables:
        temperature_value: float # temperature to set in Celsius
      then:
        - climate.mhi.set_external_room_temperature:
            temperature: !lambda "return temperature_value;"

# Version 4.0

MhiAcCtrl:
  # Only 20 (legacy) or 33 (includes 3D auto and vertical vanes) possible.
  # If you encounter mhi_ac_ctrl_core.loop error: -2 errors, change the frame_size to 20
  frame_size: 33
  initial_vertical_vanes_position: 5
  initial_horizontal_vanes_position: 8
  # Update the following to change the default room temp timeout
  external_temperature_sensor: room_temperature_sensor

button:
  - platform: restart
    name: Restart
    entity_category: diagnostic

climate:
  - platform: MhiAcCtrl
    name: "${devicename}"
    visual:
      temperature_step:
        target_temperature: 0.5
        current_temperature: 0.1

time:
  - platform: homeassistant
    id: homeassistant_time

binary_sensor:
  - platform: MhiAcCtrl
    defrost:
      name: "Defrost"
    vanes_3d_auto_enabled:
      name: "3D Auto"

sensor:
  - platform: homeassistant
    id: room_temperature_sensor
    entity_id: sensor.zolder_temperature
  - platform: MhiAcCtrl
    outdoor_temperature:
      name: "Outdoor temperature"
    return_air_temperature:
      name: "Return air temperature"
    outdoor_unit_fan_speed:
      name: "Outdoor unit fan speed"
    indoor_unit_fan_speed:
      name: "Indoor unit fan speed"
    compressor_frequency:
      name: "Compressor frequency"
    indoor_unit_total_run_time:
      name: "Indoor unit total run time"
    compressor_total_run_time:
      name: "Compressor total run time"
    current_power:
      name: "Current power"
    energy_used:
      name: "Energy used"
    indoor_unit_thi_r1:
      name: "Indoor (U-bend) HE temp 1"
    indoor_unit_thi_r2:
      name: "Indoor (capillary) HE temp 2"
    indoor_unit_thi_r3:
      name: "Indoor (suction header) HE temp 3"
    outdoor_unit_tho_r1:
      name: "Outdoor HE temp"
    outdoor_unit_expansion_valve:
      name: "Outdoor unit exp. valve"
    outdoor_unit_discharge_pipe:
      name: "Outdoor unit discharge pipe"
    outdoor_unit_discharge_pipe_super_heat:
      name: "Outdoor unit discharge pipe super heat"
    protection_state_number:
      name: "Compressor protection code"
    error_code:
      name: "Error code"
    vanes_pos:
      name: "Vanes"
    vanesLR_pos:
      name: "Vanes Left/Right"

text_sensor:
  - platform: MhiAcCtrl
    protection_state:
      name: "Compressor protection status"

select:
  - platform: MhiAcCtrl
    vertical_vanes:
      name: Fan Control Up Down
    horizontal_vanes:
      name: Fan Control Left Right

switch:
  - platform: MhiAcCtrl
    vanes_3d_auto:
      name: "3D Auto"

RobertJansen1, 2025-07-08:
as i understand correctly the external temp was set to the climate component as the temperature before v4.0?

Alfagek, 2025-07-08:
Correct 😉

Alfagek, 2025-07-16:
@RobertJansen1 

Thanks for the explanation and check. Now I know where the difference come from. 

If the AC indeed only works with full degrees like you say, it’s insane to make a lot off work for only .2 difference on the climate sensor in HA.