ginkage/MHI-AC-Ctrl-ESPHome #150: Vertical blade commands don't work (v3.0)

PuckStar, 2025-05-18:
What I found out is that vertical blade / plane commands like Swing, don't work.
Horizontal commands work fine and when doing it via the remote control it also works fine.
But through Home Assistant and this addon, it doesn't.

Do others have the same problem?
Any idea how to solve it?

PuckStar, 2025-06-22:
No one else has this issue? For others this works fine?

ajpalm, 2025-06-23:
How old are your units? From the FAQ vane control only works for the newer units.

My units use 'legacy' mode, so no vane control for me.

PuckStar, 2025-06-29:
> How old are your units? From the FAQ vane control only works for the newer units.
> 
> My units use 'legacy' mode, so no vane control for me.

Your comment lead me to look again at the code and in the new 4.0 code it mentions more clearly that frame buffer 20 (legacy) doesn't support vane and 3D control.
And then I realized that although I have new units, I changed the frame buffer to 20 because I had errors when using the default 33.

Now with the 4.0 code I thought, let me just try it with frame buffer 33 now and now the flashing worked fine and also I have all controls working now!!!

closed 2025-06-29 by PuckStar