ginkage/MHI-AC-Ctrl-ESPHome #152: Target temperature 16°C

290989, 2025-06-23:
Hello,

I have used the mhi cloud intgeration until today.
Now I have changed to mhi-ac-ctrl-ESPHome. 
Can I set the target temperature to 16°C (degrees)? 
That was possible on the cloud integration.

Thanks a lot.

RobertJansen1, 2025-06-23:
low temps for heating and high temps for cooling are currently not supported unfortunattely, and are on my whish list of things to still do, same goes for https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/88 . the MHI-AC-Ctrl project worked-around this by updating the internal sensor with an adjusted temperature. i still need to find time to do this but find myself not having a lot quite often :(

Marijn1250, 2025-07-02:
Ok, with some help from AI I updated the automation:

`
alias: Airco Garage – push spoofed room temp
triggers:
  - seconds: "0"
    trigger: time_pattern
actions:
  - variables:
      real: "{{ states('sensor.s20_garage_ds18b20_temperature') | float(nan) }}"
  - choose:
      - conditions:
          - condition: template
            value_template: "{{ real == real }}"
        sequence:
          - data:
              value: "{{ real + 5 }}"
            action: esphome.airco_garage_set_api_room_temperature
    default:
      - data:
          name: Airco Garage
          message: Skipped spoof – sensor unavailable
        action: logbook.log
mode: single

`