ginkage/MHI-AC-Ctrl-ESPHome #155: update keeps failing

TGgrimreaper, 2025-07-14:
Hi 

Im proberly an idiot but

when updating to the latest version i get the following error:
Failed config

esphome: [source /config/esphome/airco zolder.yaml:9]
  name: esphome_web_89435e
  friendly_name: Airco Woonkamer
  
  Please remove the `platform` key from the [esphome] block and use the correct platform component. This style of configuration has now been removed.
  platform: ESP8266
  board: d1_mini
  platformio_options: 
    board_build.f_cpu: 160000000L
  includes: 
    - mhi_ac_ctrl.h
    - MHI-AC-Ctrl-core.h
    - MHI-AC-Ctrl-core.cpp

`# Version 2.1
substitutions:
  # Unique device ID in HA
  deviceid: "esphome_web_89435e_mhi_ac_ctrl"
  # Unique device name in HA
  devicename: "airco_zolder"

esphome:
  name: esphome_web_89435e
  friendly_name: Airco Zolder
  platform: ESP8266
  board: d1_mini
  platformio_options:
    # Run CPU at 160Mhz to fix mhi_ac_ctrl_core.loop error: -2
    board_build.f_cpu: 160000000L
  includes:
    - mhi_ac_ctrl.h
    - MHI-AC-Ctrl-core.h
    - MHI-AC-Ctrl-core.cpp

packages:
  # Comment if you encounter mhi_ac_ctrl_core.loop error: -2 errors and uncomment the legacy_framesize file
  mhi-ac-ctrl: !include conf/large_framesize.yaml
  # Uncomment this to allow for a working configuration on floor units or older units
  # mhi-ac-ctrl: !include conf/legacy_framesize.yaml

wifi:
  networks:
    - ssid: "**********"
      password: "************"
    - ssid: "**********"
      password: "***********"
  ap:
    ssid: ${devicename}

logger:
  level: DEBUG
  baud_rate: 0

ota:
  platform: esphome

button:
  - platform: restart
    name: Restart
    entity_category: diagnostic

api:
  reboot_timeout: 0s
  services:
    # Call the set_api_room_temperature service from HA to override the room temperature
    # If a new value has not been received after room_temp_api_timeout seconds, it will fall back to internal sensor
    - service: set_api_room_temperature
      variables:
        value: float
      then:
        - lambda: |-
            return ((MhiAcCtrl*)id(${deviceid}))->set_room_temperature(value);
    # Call the set_vanes service from HA to set the vane position
    # Needed because the ESPHome Climate class does not support this natively
    # Possible values: 1-4: static positions, 5: swing, 0: unknown
    - service: set_vanes
      variables:
        value: int
      then:
        - lambda: |-
            return ((MhiAcCtrl*)id(${deviceid}))->set_vanes(value);

sensor:
  - platform: uptime
    name: Uptime
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s

text_sensor:
  - platform: version
    name: ESPHome Version
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID

`

Can some one help me update to the latest version?

TGgrimreaper, 2025-07-20:
Updated the yaml to the latest format. Still at 4 12 24 and no update option

RobertJansen1, 2025-07-20:
@TGgrimreaper did it work now or do you still have issues?
@ervee thanks for the compliments :) much appreciated but do note there are other contriutors as well!

TGgrimreaper, 2025-07-20:
the one that stays at 2024

`substitutions:
  deviceid: "esphome_web_89435e_mhi_ac_ctrl"
  devicename: "airco-zolder"

esphome:
  name: airco-zolder
  friendly_name: airco-zolder
  min_version: 2024.6.0
  platformio_options:
    board_build.f_cpu: 160000000L

esp8266:
  board: d1_mini

logger:
  level: INFO
  baud_rate: 0

ota:
  platform: esphome
  password: "**********"

wifi:
  networks:
    - ssid: "**********"
      password: "************"
    - ssid: "*************"
      password: "**************"
  ap:
    ssid: "****************"
    password: "****************"

captive_portal:

external_components:
  - source: github://ginkage/MHI-AC-Ctrl-ESPHome@master
    components: [MhiAcCtrl]

api:
  encryption:
    key: "***********************************"
  reboot_timeout: 0s
  services:
    - service: set_external_room_temperature
      variables:
        temperature_value: float
      then:
        - climate.mhi.set_external_room_temperature:
            temperature: !lambda 'return temperature_value;'

MhiAcCtrl:
  frame_size: 33
  sck_pin: 14
  mosi_pin: 13
  miso_pin: 12
  initial_vertical_vanes_position: 5
  initial_horizontal_vanes_position: 8
  room_temp_timeout: 60

button:
  - platform: restart
    name: "Restart"
    entity_category: diagnostic

climate:
  - platform: MhiAcCtrl
    name: "MHI Air Conditioner"
    temperature_offset: true
[... 231 lines cut ...]
  - platform: wifi_info
    ip_address:
      name: "IP"
    ssid:
      name: "SSID"
    bssid:
      name: "BSSID"
  - platform: MhiAcCtrl
    protection_state:
      name: "Compressor protection status"

select:
  - platform: MhiAcCtrl
    vertical_vanes:
      name: "Fan Control Up Down"
    horizontal_vanes:
      name: "Fan Control Left Right"
    fan_speed:
      name: "Fan Speed"

switch:
  - platform: MhiAcCtrl
    vanes_3d_auto:
      name: "3D Auto"
`

RobertJansen1, 2025-07-20:
Can you paste the log when you click update and can you use  ```` tags for readability

TGgrimreaper, 2025-07-20:
the code button gave me ' '

ill try it
''INFO ESPHome 2025.7.2
INFO Reading configuration /config/esphome/airco zolder.yaml...
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing airco-zolder (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP @ 2.0.0
|-- ESP8266WiFi @ 1.0
|-- ESPAsyncWebServer @ 3.7.10
|-- DNSServer @ 1.1.1
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.10
RAM:   [=====     ]  46.0% (used 37688 bytes from 81920 bytes)
Flash: [=====     ]  53.2% (used 555469 bytes from 1044464 bytes)
========================= [SUCCESS] Took 10.07 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of airco-zolder.local in mDNS
INFO Resolving IP address of airco-zolder.local
ERROR Error resolving IP address of airco-zolder.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname''

i dont have update, only install

gabcraftia, 2025-07-20:
> [@gabcraftia](https://github.com/gabcraftia) we can update the readme with a FaQ on this, can you make a merge request for this or do you have a copy of the error message so we can add it easily?

unfortunately I don't have a exact copy of it other then the screenshot.

TGgrimreaper, 2025-07-20:
Weird in dont understand, i have set an static ip and its working so i dont get how its not connected. The other wifi unit works flawless with the update to the new platform and this one is a pita

TGgrimreaper, 2025-07-20:
unable to acces the module w/o opening the unit and no web interface... :( :( on neither units. pfffff

RobertJansen1, 2025-07-20:
If I am not mistaken, configuring the manual ip in the yaml will automatically use that ip to connect to the unit and do the ota. Follow the link from the error message and you will see how it can be configured. 

This has to do with dns/mdns in your network and potentially has to do with a different ap?

If home assistant can still control the unit, ota should also be possible without opening the unit

TGgrimreaper, 2025-07-20:
yea no isseus  with controlling. It just keep jabbing about no network connection wilst it clearly is connected. I do use a mesh network and this esp home module is on a other AP?

closed 2025-07-22 by TGgrimreaper