ginkage/MHI-AC-Ctrl-ESPHome #166: idea: Silent mode detection/activation

SuperKoe, 2025-08-14:
On the remote of the MHI (SKxx ZS-W) is a button Silent.
This makes the outside unit more silent, by lowering the FAN and COMPRESSOR

The SC-BIKN2-E kit with RC-EX3 (MHI external thermostat) you can set these values, it also uses the same connector as the MHI-AC-Ctrl-ESPHome
Datasheet: [SC-BIKN2-E](https://mitsubishi.kh.ua/wp-content/uploads/2021/10/18-SRK-T-265.pdf)(page 10 mentions silent mode) 

REMOTE CONTROL manual: [RC-EX3 ](https://www.mhi-mth.co.jp/en/products/pdf/pjz012a131_english.pdf) (page 40)

Mode of silent on the MHI AC:
[image]

Is it possible to detect silent mode and activate it with MHI-AC-Ctrl-ESPHome?

RobertJansen1, 2025-08-15:
Can you try it? (Or ask chatgpt where it got this knowledge from)

SuperKoe, 2025-08-15:
> silent mode was researched before and then said to not be in the new wifi module: [absalom-muc/MHI-AC-Ctrl#77 (comment)](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/77#issuecomment-1741991290)
> 
> if we can snif the data from this module using https://github.com/absalom-muc/MHI-AC-Trace/tree/main we can try and implement reading and writing silent mode
> 
> do you own a SC-BIKN2-E kit with RC-EX3 [@SuperKoe](https://github.com/SuperKoe)

No i dont.

Using silent on the (default) remote works, and is persistent.
I turned on the airco today via Home assistant and the silent mode is still acivated (is see it on the compressor rpm and fanspeed).

In the manual of the SC-BIKN2-E clearly states the silent mode can be activated, so in theory you can sniff it out? but then you need a SC-BIKN2-E (€ 150 ugh)

RobertJansen1, 2025-08-15:
> No i dont.
>
> Using silent on the (default) remote works, and is persistent. I turned on the airco today via Home assistant and the silent mode is still acivated (is see it on the compressor rpm and [...]

In theory it is sniffable. The cost will be around €320 for the SC-BIKN2-E kit with RC-EX3

arpiecodes, 2025-08-15:
Isn't the remote control using a whole different bus/communication protocol from the CNI port? Obviously you can hook up a logic analyser and try to sniff/decode that protocol too, but I don't think it is related in any way to the WiFi module's protocol.

Also, the fact that setting 'sticks' after setting it through other means kind of confirms for me these bits are not available on the SPI bus. If they were, then this setting would be overwritten every time the WiFI module/MHI-AC-Ctrl sends through an update control message (like the extra vane positions that were added in the extended message being overwritten before we had support for the extended message bits).

ajpalm, 2025-08-15:
> Using silent on the (default) remote works, and is persistent.

Intesting, as that is different behavior to how the cooling cleaning cycle works.

> If they were, then this setting would be overwritten every time the WiFI module/MHI-AC-Ctrl sends through an update control message

This is what happens with the cooling cleaning cycle. If a 'remote' that doesn't have cleaning enabled sends a command, it is turned off.

SuperKoe, 2025-08-16:
> Isn't the remote control using a whole different bus/communication protocol from the CNI port? I'd expect it to be cause SPI is very difficult to extend in terms of cable length for more than a couple of tens of cm's.
> 
> Obviously you can hook up a logic analyser and try to sniff/decode that protocol too, but I don't think it is related in any way to the WiFi module's protocol. It'd require a lot of effort.
> 
> Also, the fact that setting 'sticks' after setting it through other means kind of confirms for me these bits are not available on the SPI bus. If they were, then this setting would be overwritten every time the WiFI module/MHI-AC-Ctrl sends through an update control message (like the extra vane positions that were added in the extended message being overwritten before we had support for the extended message bits).
> 
> But; this is largely "source: trust me bro". :)

There is indeed little info, but maybe its easy; the connector we all love; is also called SC-BIKN

[image]

"source: trust me bro". :) hihi

TobiasB71, 2025-12-10:
> There is a converter box that does the translation from spi to something else for the remote (or modbus if you buy a modbus adapter) I am considering buying a remote with converter box but I need about 100 euro's to do so, if someone wants to help me with paying for the experiment please use the PayPal donate button
> 

It appears the guys behind ClimateControl from ProtoArt have figured it out as their latest firmwares now support setting the Silent mode.

How far are you on the donation side, ie. how much is missing in order to proceed?

mreijnde, 2026-09-07:
I recently had a Mitsubishi Heavy Industries AC system installed (SCM60ZS-W with 3 × SRKxxZS-W indoor units) and added [mhi-ac-ctrl-esp32](https://github.com/hberntsen/mhi-ac-ctrl-esp32) to control them. One feature I was missing was Silent Mode, so last weekend I decided to see if I could figure it out. Good news: I got it working on my system!

To obtain the protocol information, I bought a ProtoART Smart Climate Control. A nice product, by the way, definitely recommended if you want this functionality without doing any tweaking yourself.

I traced the SPI communication while enabling and disabling Silent Mode and, based on those traces, I think I now understand how the feature is represented and controlled. I documented my findings in an updated [SPI protocol document](https://github.com/mreijnde/mhi-ac-ctrl-esp32/blob/master_modified/SPI_protocol.md).

I also created a [modified version of mhi-ac-ctrl-esp32](https://github.com/mreijnde/mhi-ac-ctrl-esp32/tree/master_modified) on my github. It currently adds:
- Silent Mode switch
- SPI logging, which enables dumping MOSI/MISO traffic to the ESPHome log for further protocol analysis

On my SCM60ZS-W setup, Silent Mode is working so far with the ESP32 version, but I haven't tested it for an extended period yet.  I haven't connected an ESP8266 yet, so I haven't ported the changes to MHI-AC-Ctrl-ESPHome itself yet.

It would be great if someone with a different MHI unit could test the [modified mhi-ac-ctrl-esp32 version](https://github.com/mreijnde/mhi-ac-ctrl-esp32/tree/master_modified) and report whether Silent Mode works. That would help confirm whether the protocol bits I found are consistent across different models.

Any testing, SPI traces, or feedback would be very welcome!

Pel1can111, 2026-09-08:
Excellent work, I have 3 indoor units using ESP8266 so would be happy to test on those if it did get ported over.

Out of curiosity, do you notice much performance impact or change in power consumption from running it in silent mode?

mreijnde, 2026-09-09:
The silent mode is using I think about half of the electric power and is more silent. The max compressor speed and max fan speed of the outdoor speed become lower.  I want to also use it to optimize that AC is more using my solar panel energy directly. A more constant (less modulation) but lower power usage typical helps for that.

mreijnde, 2026-09-15:
@maxhov: nice that you already implemented the findings in the esp8266 firmware.

In parallel @hberntsen also immediately made an improved implementation for the esp32 firmware.
https://github.com/hberntsen/mhi-ac-ctrl-esp32/pull/42

Some new insights:
- It looks like DB6 is not required to set the silent mode
- It looks like you cannot override silent mode set by IR remote or by another indoor unit (in case of a multi-split unit).  This need more testing.